dzlonline/the_synth #3: Only outputs Saw Wave

aframires, 2016-06-14:
Hello everyone! 
I'm trying to use this synth in an arduino UNO and, even in all examples, the only output I have in the osciloscope is a saw wave! I'm loading examples using the suggested connection:
//                    +10µF 
//PIN 11 ---[ 1k ]--+---||--->> Audio out
//                  |
//                 === 10nF
//                  |
//                 GND

Anyone knows what's the problem?

aframires, 2016-06-14:
[image]

troisiemetype, 2016-06-14:
Did you try to listen to it, or just saw it? Is your setting ok? It loks like the charge/discharge of the capacitor, but with various decharging slopes.
Also, I can see that the frequency setting is 62,5kHz, so there's a chance you're looking to the PWM that generates the audio wave. What do you see when setting a frewuency around what you should hear? That's to say, something between around 100-10000Hz.

dzlonline, 2016-06-14:
Triangle is just a PWM artifact. Try listening to the signal (looks to be
correctly modulated)

aframires, 2016-06-14:
I haven't explained myself correctely. The output wave was supposed to be a sine wave, in this situation  we are using example "modulation.ini". I'm measuring in the audio output node.

dzlonline, 2016-06-14:
Yes :)