dzlonline/the_synth #7: How to stop a voice?

arpruss, 2017-07-11:
I have a really stupid question: I can set the note for a voice with mTrigger(), but I don't see an API to stop one particular voice. Am I missing something, or is the API missing?

dzlonline, 2017-07-11:
Heh, that's right didn't implement that, I guess I never had use for it...
I would briefly set the length of the voice to 0 which would cause instant
decay.
I guess you could implement a note-off this way...

If you need to mute/unmute the synth use .suspend() and .resume()

I will consider a voice.off() in the future lib.

Regards Dzl.

arpruss, 2017-07-12:
setLength(voice,0) does the job.
By the way, what units is the length in?

perremba, 2020-04-26:
You might use suspend in order to stop the interrupt, and then reconfigure the pin as input (to cut the remaining noise)