dzlonline/the_synth #10: Can't get it to work with mega2560

13um45, 2018-06-15:
I've been using this without issue on my uno. Trying to get it to work with a mega2560 but sound isn't being output. I read in this issue: https://github.com/dzlonline/the_synth/issues/6 that you just need to connect to pin 9 or 10, neither seem to work for me. Any ideas as to why or how you would port this for a mega? Thanks

troisiemetype, 2018-06-15:
Hi,

It should work as is, the register definitions are the same between Atmega 328P and 2560.
On page 55 of the datasheet it's written that power reduction register has to be properly set to enable timer 2. Although I think the Arduino environnement takes care of that, maybe it's worth a try?

nathanRamaNoodles, 2018-06-27:
I also had the same issue.  I thought my Mega was broken, so I tested if my timer 2 was broken, but it wasn't.  I think this library needs to use #if defines for the Mega and the Uno.  Hopefully, this will encourage this library to also work on the esp8266.