dzlonline/the_synth #15: Thank you

nathanRamaNoodles, 2018-09-13:
I played around with your awesome library and created my own.  Just thought I'd share it with you:
https://github.com/nathanRamaNoodles/MusicWithoutDelay-LIbrary

nathanRamaNoodles, 2018-10-05:
Yup, after a couple of months, I figured it out.  It was quite the challenge because pin 11 on Mega was broken since 2011 after Arduino updated their IDE.  So after trial and error, and a little help from [PaulStoffregen](https://github.com/PaulStoffregen), I figured out the solution.  It was a simple analogWrite( ) along with the 16-bit pre-scaler.  and it sounds beautiful.  I even got carried away and increased polyphony to infinity([well there are limitations](https://github.com/nathanRamaNoodles/MusicWithoutDelay-LIbrary#compatibility)).  If you want to use the synth engine you can simply extract the synth.h and synth.cpp files and create a library out of that.  I simply used this synth "engine" in order to play music without delay.

I wish more people would find my library, but oh well.

13um45, 2018-10-05:
Amazing! I've been stuck on a musical cellular automaton project for a while. This completely unblocks that, thanks you!