dzlonline/the_synth #19: Synth with barebone ATMEGA328p PU chip

montythepython, 2020-10-05:
Hi @dzlonline, 

Thank you for sharing this awesome project!
I have been using your library without any issues on different nano boards. I recently got my hands on a couple of atmega328p PU (28DIP) chips. 

After boot loading and uploading the sketch it all works well until I connect the audio out pin (in my case, pin 3, using CHB).

I know the sketch is running well without the audio input connected because the code runs an LED sequence (sequencer steps) and it works. I modify the tempo with a delay function in the sketch and turning the tempo potentiometer affects the LED sequence tempo accordingly.

Moreover, sometimes I connect the audio pin out and I do get some notes. Sometimes just one and it freezes, sometimes a full sequence and then it freezes...

I have tried different chips and get the same result. I was wondering if this can be related to the way the Synth.h is setting up the board.

@troisiemetype, reading the reported issues before creating one, I found a thread with you that made me think you can also have a good idea about what is happening here.

Thanks!

dzlonline, 2020-10-05:
Hi.
Question: by "connect the audio out pin" do you mean physically?
Dzl

montythepython, 2020-10-06:
Hi dzl,

Yes when I connect the speaker to pin 3 and ground.

Or I can have pin 3 and ground wires to the audio jack and when I plug in
the speaker/headphones, I have the issues I described

Thank you!

dzlonline, 2020-10-07:
Hi,

Sounds like your problems are electrical.
If you connect a speaker to a single pin (e.g. between CHA and GND) you
need to put a DC blocking capacitor in series with the speaker: e.g. 220
microfarads with the positive end connected to CHA and negative end to the
speaker.
If you do not want to use a capacitor you can use the DIFF mode and connect
the speaker between CHA and CHB.

Let me know if it solves your problem.
Cheers
Dzl

montythepython, 2020-10-07:
Hi, thank you.

Ok I will try it tonight. I have two set ups at the moment. Same circuit on
both, one with a nano and one just with the chip (plus crystal oscillator,
etc)

At the beginning I had nothing between the speaker and CHA but like this,
it does work on the nano and not on the chip.

I continued working on the circuit with the nano and added what I thought
was a low pass circuit you show in the beginning of your code examples but
I haven't tried it on the circuit with the chip.

Thank you, I will let you know.

Cheers

montythepython, 2020-10-08:
Hi dzl!

That was it! Thank you. No idea why it works with the nano but not with the
chip...

Next step is adding a midi out and integrating the code to what I already
have for a sequencer done with a 4051 chip.

Nice blog dzl, very impressive!. Looking forward to playing with your synth
library for the ESP32!

Cheers