dzlonline/the_synth #20: Made it work on the Leonardo, are you interested in the code?

Frazatto, 2020-10-23:
Don't know if you are interested, but I got in touch with user df99 that made a comment at your Instructable about this project and made a modification so it will work correctly using the Uno or Leonardo.

Are you interested?

dzlonline, 2020-10-29:
Hi André

Absolutely

Regards
Dzl

Frazatto, 2020-10-30:
[synth.h.tar.gz](https://github.com/dzlonline/the_synth/files/5465796/synth.h.tar.gz)

Nice to meet you Dzl  o/

    Talking to df99 and when I sent you that message, I understood he had tested all his modifications and everything worked as intended.
    But apparently he tested only the cases he was using in his project (diff mode pins 5 and 13).
    When I tried to use just pin 10 on my project, it apparently also changed pin 9 (the complementary pin, which was already being used for a shift register). I'm now using pin 5 successfully, but there is no accessible pin 13 on the ProMicro (a small Leonardo) to be sure it actually is just changing pin 5 behavior.

    I'm a computer engineer, so I'm very comfortable coding, not so much handling hardware.
    I don't have an oscilloscope and it's hard to try debugging pwm signals with a cheap multimeter.
    I tried my luck with the chipset documentation and my lack of experience with low level operations let me hanging.

    So now I"m not sure this is very helpful to you as is anymore......
    I think the way I write in english (I'm from Brasil  o/) didn't help and he insisted his modifications worked flawless.
    If you'd like to give it a try, this is the  atmega16u4-32u4 datasheet.
    [](http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-7766-8-bit-AVR-ATmega16U4-32U4_Datasheet.pdf)

    What I discovered is that if you comment out line 174 of the lib, there is no sound but it stops the erratic behavior. You can find that register at page 164 of the documentation and what I understood it's where the pin is selected to "connect" to the pwm signal for group B. It makes sense, but I quickly got lost with all the bit options and what they ACTUALLY do, I may be wrong all together.
    It most definitely works on both diff modes though! I tested in another board with a small speaker.
 
    Anyway, thanks for this lib man, it will make my project much more interesting. Hope I can contribute a little to it!
    Would you mind if I ask you a bunch of stuff on the next days?
    If you don't mind, you can send me e-mails directly.
    I'm hacking a yamaha toy keyboard, already made it work as an MIDI Controller and now i'm trying to add proper synth functions to keep it working as a keyboard also.
    [](https://www.instagram.com/p/CGsxWfggD3x/?utm_source=ig_web_copy_link)