dzlonline/the_synth #21: Great library! Couple tweaks suggested

prosper00, 2020-11-25:
I just got this working on my project - it's a sequencer that constructs and plays chords and chord progressions, based on another project I found online. ( https://youtu.be/bd3h9vR5jO8 )

https://github.com/prosper00/arpeggiator - it's messy spaghetti code at present, but it does work.

I made a couple minor changes, to get it to cooperate with other stuff going on on the CPU. 
1. Change the ISR definition to "ISR(TIMER1_COMPA_vect,ISR_NOBLOCK)" - the NOBLOCK allows the isr itself to be interrupted, which lets me do some UI stuff like reading analogs in my own ISR, and respond to user inputs while playing. As long as I keep my isr's short, this doesn't cause any glitches.
2. Replace the 16000000.0 literals with the F_CPU defines, so that I can run at clock speeds other than 16M
3. Change the data types in tables.h to eliminate the type conversion warnings

dzlonline, 2020-11-25:
Hi.
Thanks for the suggestions.
I'm currently working on a fully scalable/portable version of the lib. so
changes may be reflected there.

Regards
Dzl

prosper00, 2020-11-25:
excellent! I look forward to it! I was considering porting it over to the STM32 for my project - more IO's and cpu cycles would be useful, and IIRC, the STM32F4 has an FPU