dzlonline/the_synth #22: Ported to Arduino DUE

zircothc, 2021-03-17:
Hi.
Sorry, this is not an issue at all.
Some time ago I ported your library to run with Arduino DUE.

Years ago before that I changed your library to be able to use a 12 bit spi dac with arduino nano. I build  a synth/sequencer with that and used it a lot.
Due has more power, tones of GPIO and on board dual DAC so I tried it and It was quite easy. 
With DUE I used "DueTimer" library to call audio driver interrupt and direct dac write to put samples out.

I added more tables and even a filter from mozzi library. I got at least 16 voice polyphony and built a drum machine: some pots, two RGB matrix leds as screen and a resistive touch panel on top. Sound is amazing and very fun to play. I was very happy with the result.

If you want I can share the library with DUE support.

Also try to port to esp32 and I2S DAC but sound is not as good as I got with DUE. My skills are not as good as I wish :(

I saw you were developing a library for ESP32 in your blog but still no code.
Can you share it? It would be fantastic ;)
Thank you

FluxGarage, 2021-03-19:
> If you want I can share the library with DUE support.

Hi Carlos, I'd love to test your adjusted library on the Due. Please share it! :)

zircothc, 2021-03-22:
Hi.
Sorry for delay.
This attached version is as the original but audio output adapted to DUE. It has 4 voice poly and no other improvements.
Higher polyphony, more tables, vol parameter, filter, ... are not included. I Need time to clean my lib and adapt again new version of mozzi lib to use the filter.

[synthDUEtest.zip](https://github.com/dzlonline/the_synth/files/6182014/synthDUEtest.zip)

Hope it helps.
;)

dzlonline, 2021-03-23:
Hi Carlos.

Thanks for the adaption.
I Will incorporate it as soon as I get hold of a DUE :)

Cheers
Dzl

FluxGarage, 2021-03-23:
Hi Carlos, 

thank you for sharing your library, I'll test it soon! :)

FluxGarage, 2021-03-29:
Hi Carlos,

I tested your adaption on an Arduino Due (Clone) and it works pretty fine! It's a great thing to use this synth library while still having so much horse power left.
It should be mentioned that one has to include the "tables.h" from the original library in order to compile the sketch and that the audio connection has to be made on pins DAC0, DAC1 and GND.

I'm curious about your experiments with higher polyphony ;)

Big thanks again!
Dennis

zircothc, 2021-03-29:
Hi Dennis.
To add polyphony i've changed all arrays from 4 elements to 16 and at the BUILD_SOUND isr add that voices to the output.
Attached version has 16 bit poly. Also added **tables.h**. It has a few more tables, 0 to 14
I use 8+8 voices, 8 voices per DAC. You can uncomment that part and use 16 poly with the same output or do what you like.
Also included VOL parameter so don't forget to add that new parameter in setupvoice calls: ("7" at this example)
**edgar.setupVoice(0,SINE,60,ENVELOPE0,80,64, 7);**
VOL parameter must be between 0 and 12. More than that will break your speakers or headphones or ears at all ;)

[the_synthdacdue2021.zip](https://github.com/dzlonline/the_synth/files/6221338/the_synthdacdue2021.zip)

On the other hand, don't forget a pair of resistor (i'm using 680ohm) attached on each DAC pin. 

I Hope you can test it.

Also I hope @dzlonline can share soon ESP32 lib ;)

A short demo:
https://user-images.githubusercontent.com/17828930/112833048-735e8300-9096-11eb-809a-9873795d9aa7.mp4

zircothc, 2021-03-29:
Excuse me. I forgot to tell you that there is no filter in that version :(
Filter is part of a modded mozzi library. I need time to mod new mozzi library version (the last one has ESP32 support) to be compatible with DUE again.

FluxGarage, 2021-04-01:
Hi Carlos,

thank you for the detailed instructions and the polyphony version. I think I'll play around this next weekend :)

Regarding the video:
That's a really cool machine that you've build there! Is that a selfmade touch/led grid? 
Maybe that's a cool thing for instructables.com ?

zircothc, 2021-04-01:
Hi. 
Glad you like It.
Dzl library is perfect to build electronic drum sounds and very easy to program with.

That drum machine project is on "resume" mode.
The touch panel is a 8" resistive one you can buy for about 8€. Is a 4 wire. very easy to read, but not enough accurate for this task.
I got a capacitive touch panel, gt911, I2c protocol. Costs the same.
 I got It to work :) but I'm still mixing code to use It.
Display is a pair of 8x8 RGB led Matrix yet made. They are put inside a 3d printed frame to show 8x4 cells, each one with 2x2 leds :)
So I got a 8x4 virtual buttons.
Then It is easy to map a touched cell into a led cell. Is a very fast system for developing control panels.
Hope i get time to finish It.
;)