dzlonline/the_synth #24: Noise in speaker when 2 arduinos connected via Serial pins

FE3DS, 2021-09-28:
Hi, very good project!
I use your synth with arduino Pro-mini 5V/16MHz and keypad to generate DTMF tones.
My Pro-Mini is connected to a MEGA2560 via Serial port.
But I have a problem:
If MEGA and Pro-Mini powered by their own power supply and connected each other via GND/TX/RX pins -> OK
If MEGA and Pro-Mini powered by the same power supply and connected each other via 5V/GND/TX/RX pins -> no OK. I have a noise in speaker.
How can I resolve that?
Thanks in advance.

dzlonline, 2021-09-28:
Hi.
Sounds like a ground-loop/power supply problem. Try connecting all ground
wires at one central point near the power supply. Try adding a large
capacitor ~1000µF between this ground point and the 5Vsupply (also all
connected in one point). You may try to pass all serial signals via. 100R
resistors to limit RF noise and fast currents.

The speakers (if connected directly to the chip) can draw around 100mA AC
current at 32KHz this is likely transmitted via. the 5V line.

Let me know how it goes...
Cheers Dzl

FE3DS, 2021-09-28:
Hi,
Thanks for these informations.
I will purchase this capacitor and test that.
Thanks a lot.

Bien cordialement,
Nicolas BOULAY.
FE3DS.

www.fe3ds.fr<http://www.fe3ds.fr/> -- www.facebook.com/FE3DS<http://www.facebook.com/FE3DS>

dzlonline, 2021-09-29:
For testing you can also just run one of the Arduinos from a power bank and
_not_ connect 5V between the boards. This should have the same effect.