dzlonline/the_synth #27: license of repo

Fearge, 2023-10-28:
Hi,

this is not really an issue but I couldn't find these information anywhere.
I'm currently studying media technology in second semester and we have a class where we build a synth.
We would like to use your project as base and modify it to our needs. Since you don't have any license files attached I want to make sure that this is fine to you.
The project won't be distributed or published, it's just an internal project.
It would be great if you could let me know if this use of your code is okay for you.

Thanks in advance
Greetings from Germany

dzlonline, 2023-10-28:
Hi.
You hereby have the permission to use/modify the code as you discribe :)
Let me know if you have any questions otherwise.
Regards
Dzl
(Should probably get around to add the license)

Fearge, 2023-10-28:
Great, Thanks a Lot! Now I'm sure my professor wont't have any problems with that